fossasia/susi_linux #403: To have a static ip address on SUSI Linux

stealthanthrax, 2018-10-18:
**Describe the bug**
When SUSI Linux is operating in Normal mode, connected to a wifi, it has a fluctuating IP address and connecting via SSH connection makes it a tedious task.

**To Reproduce**
Steps to reproduce the behavior:
1. Complete the SUSI installation
2. Start SUSI
3. Try to connect it via SSH
4. See error

**Expected behavior**
To have a static IP address and improve the process of connecting  to the Pi via SSH connection

**Additional context**
Id like to work on this

hongquan, 2018-10-19:
If someone wants to SSH to RPI without trouble of fluctuating IP address, he/she should either:

- Use RPi's mDNS domain name `raspberrypi.local` instead of IP address. Linux laptop should be installed these packages `libnss-mdns`, `avahi-daemon`, which are already pre-included in major distros (Ubuntu, Fedora).

- Configure his/her home router to assign fix IP for RPi (based on MAC address).

**Note**: Don't set static IP _in RPi_ itself, which makes RPi not portable.

Explanation: If you set static IP `192.168.1.10`, for example, in RPi (_/etc/network/interfaces_ file), that RPi will not be usable when moved to other LAN network, because:

- New network may be in different subnet, `192.168.0.x`, for example.
- The RPi's static IP may be conflict with an IP managed by that network's DHCP server. DHCP server doesn't know that RPi takes `192.168.1.10` and it tries to give that IP to another client machine and that makes _both_ machines not work.

stealthanthrax, 2018-10-19:
Okay @hongquan , makes sense. We should then add these ways in the documentation for easy access.

hongquan, 2018-10-20:
I added a wiki here: https://github.com/fossasia/susi_linux/wiki/Connect-to-RPi-with-a-fixed-address

closed 2018-10-20 by hongquan