fossasia/susi_linux #406: Integrate SUSIbian build script to Travis

hongquan, 2018-10-22:
This issue is to track what I'm doing to make it able for Travis to run our SUSIbian (Rasspbian for SUSI) build script and generate the image.

Because the build script need to install ReSpeaker driver, and the ReSpeaker install script [explicitly check](https://github.com/respeaker/seeed-voicecard/blob/master/install.sh#L8) for the presence of Raspberry Pi, our build script need to be ran on real Raspberry Pi hardware.

hongquan, 2018-10-22:
Found this Docker image: https://hub.docker.com/r/multiarch/debian-debootstrap/. Don't know if it can help something.

hongquan, 2018-10-22:
pi-gen author include a script to let run the build script inside a Docker container. This Docker container is also include QEMU, so that it can run ARM binary. I'm trying this Docker script, to see if it can generate Raspbian successfully.

hongquan, 2018-10-22:
Encounter some error related to QEMU when running the above Docker script:

```
[18:44:46] Begin /pi-gen/stage2.5/01-susi/01-run-chroot.sh
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
sudo: unable to resolve host 6b84ec72e811
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
sudo: unable to resolve host 6b84ec72e811
Cloning into 'susi_linux'...
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
sudo: unable to resolve host 6b84ec72e811
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
sudo: unable to resolve host 6b84ec72e811

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
```

hongquan, 2018-10-22:
Other error:

```
Unsupported ioctl: cmd=0xffffffff81785501
arecord: device_list:270: no soundcards found...
installing Respeaker Mic Array drivers from source
Cloning into 'seeed-voicecard'...
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
Unknown host QEMU_IFLA type: 40
Unknown host QEMU_IFLA type: 41
Unknown host QEMU_IFLA type: 47
Unknown host QEMU_IFLA type: 48
Unknown host QEMU_IFLA type: 43
sudo: unable to resolve host 6b84ec72e811
/bin/cat: /proc/device-tree/model: No such file or directory
Sorry, this drivers only works on raspberry pi
Running in chroot, ignoring request.
Running in chroot, ignoring request.
[18:47:14] End /pi-gen/stage2.5/01-susi/01-run-chroot.sh
```

stealthanthrax, 2018-10-23:
@hongquan , I am already trying to integrate this script in Travis. In my latest PR and its almost near completion . Please wait.

hongquan, 2018-10-24:
@stealthanthrax Which PR do you mean? #405 or #392 ?

stealthanthrax, 2018-10-24:
@hongquan , in #392

hongquan, 2018-10-25:
What I found so far:

- TravisCI doesn't provide real ARM hardware.
- {QEMU + x86 machine + Docker} (Docker container is to make sure things are setup correctly) can build Raspbian Lite image (without our software).
- When {QEMU + x86 machine + Docker} run our customization part (to build SUSIbian), QEMU throws some warning messages (because some system calls are not supported in QEMU). The ReSpeaker driver script refuses to work, because it detects that the machine is not Raspberry.

I think we have 2 directions to try:

1) Continue with {QEMU + x86 machine + Docker}. But we need to do:
- Patch ReSpeaker driver install script, to bypass detection phase. But I'm not sure if we bypass this phase, the later phase can still work or not.
- Rewrite our build script in a way so that it doesn't call system calls which QEMU doesn't support. How big amount of work is unknown yet.

2) Run the build script in real Raspberry Pi and find other CI service to manage it.
- The CI system must have two parts: master and worker (or agent). The worker will run in our RPI and execute our script. The master is the part that we integrate to GitHub repo, trigger the task and collect the result from worker.
- Depend on the particular CI service, it can require our RPi to be public accessible from Internet or not (so that the "master" can contact the "worker"). If it requires, we need to find where to place our RPi (I'm thinking about placing it in HotelXoai).

I find another advantage of the 2nd direction:

- We can cache part of our work, to reduce build time. To be clearer, our SUSIban = Raspbian Lite + SUSI softwares. We can cache the "Raspbian Lite" folder, so that every we trigger the build job, it only do "installing SUSI softwares". I think that this is a big advantage, because it takes several hours to build the image.

I'm research [Buildkite](https://buildkite.com).

hongquan, 2018-10-25:
Trying running build script from Buildkite. First pic is what we see from Buildkite webpage. Second pic is that _buildkite-agent_ is running the script in my RPi:

[image]

[image]

hongquan, 2018-10-26:
Build log from the last trial:

[build-susibian_build_13_run-build-dot-sh.log.gz](https://github.com/fossasia/susi_linux/files/2520120/build-susibian_build_13_run-build-dot-sh.log.gz)

Problem: Failed to build Java SUSI Server, due to `java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty`.

hongquan, 2018-10-27:
[Fixed](https://github.com/fossasia/pi-gen/commit/3d37bd4d2db99fccfdd73f24049c04dc9ebf8c25) the Java error.
The code for Buildkite integration is in https://github.com/fossasia/pi-gen/tree/susi-buildkite

hongquan, 2018-10-31:
Due to difference of repo, we want to trigger image building on event of `susi_linux` repo, but the actual build script is on `pi-gen`, we have to setup 2 Buildkite pipelines for 2 repos. The event on `susi_linux` will trigger its pipeline, which in turn triggers the `pi-gen` pipeline.

hongquan, 2018-10-31:
Successfully to make 2 chained Pipelines works.

[image]

But the build time is not satisfactory. It is too long (due to download speed not good at my home).

stealthanthrax, 2018-10-31:
I am not sure why more reconnaissance on this topic is needed. We already have an open PR which requires only some minor fixes. Team progress would be much faster if we collab on that PR.

hongquan, 2018-10-31:
@stealthanthrax What are minor fixes in #392?

How can I collab when I [requested](https://github.com/fossasia/susi_linux/pull/392#issuecomment-433595254) you to provide your build log, but you _never_ give?

I also [requested](https://github.com/fossasia/susi_linux/pull/392#discussion_r228732862) you to fix the file name. What is your response? Only silence. That is what you called "collab"?

When I have to _look for the log_ myself (without any response from you), I see that the most your script can do is to [download the kernel-qemu-4.4.34-jessie](https://api.travis-ci.org/v3/job/448981570/log.txt). And that is you called "only some minor fixes"?

stealthanthrax, 2018-10-31:
@hongquan , I updated every item in my PR description but you never read any of them.

hongquan, 2018-10-31:
@stealthanthrax Updated? Show me the links of your response of all my questions above!

I always give links as evidence. But you never!

hongquan, 2018-10-31:
@stealthanthrax:

This is your desc of that PR

> as it was the only working kernel I could find

It is not a valid explanation of "Why the OS is Debian Stretch but you use Jessie kernel".

If you don't find the right kernel, why don't you just stop the work?

stealthanthrax, 2018-10-31:
https://github.com/fossasia/susi_linux/pull/392#issuecomment-433650354

stealthanthrax, 2018-10-31:
I queried about the type of log you were refering to , but no reply was given grom your side.

hongquan, 2018-11-01:
@stealthanthrax No reply? Then what is this?

[image]

hongquan, 2018-11-12:
Solved by #410

closed 2018-11-12 by hongquan