fossasia/susi_linux #451: Support for Ubuntu

irthomasthomas, 2019-01-17:
I am trying to follow your instructions to run on ubuntu 18.04 but there appears to be information missing from the guide. For example the setup script requires parameters which are not documented. Moving past that, when I try to run the demo I am missing the alsaaudio module which is not available for python3. Am I supposed to build alsaaudio from somewhere?

Thanks in advance.

hongquan, 2019-01-18:
@irthomasthomas Sorry for outdated doc.
Recently, we put focus to Raspberry Pi, so the install script is likely not working on Ubuntu. If you have a board, please download the firmware image here: http://download.susi.ai/susibian/ and help test.

About the _pyalsaaudio_ modules. It is compatible with Python 3 (we have tested it on Raspberry Pi), but the PyPi doesn't provide a [prebuilt binary](https://pypi.org/project/pyalsaaudio/#files) for x86 yet, `pip` will have to build it from source on your machine. To help `pip` do that, you need to install some dependencies: `sudo apt install libasound2-dev`.

If you run the install script on Raspberry Pi, it doesn't have to build _pyalsaaudio_ from source, because we provide a prebuilt binary for ARM already (https://repo.fury.io/fossasia/).

irthomasthomas, 2019-01-19:
@hongquan 
Ok thanks. I followed your instructions and I'm a little further along now but now I get the error no module 'susi_python'

stealthanthrax, 2019-01-20:
@irthomasthomas , actually the code is focused on Raspebrry Pi and includes a lot of arm specific libraries. So, the code is likely to not work on x64 or x86 pcs. 
However, can you add the exact log here?

hongquan, 2019-01-21:
@irthomasthomas 

> I get the error no module 'susi_python'

The install script didn't run successfully on your machine, and hence, `susi_python` was not installed.

norbusan, 2019-04-26:
See https://raw.githubusercontent.com/fossasia/susi_installer/development/install.sh for a working version

norbusan, 2019-05-17:
Hi @irthomasthomas 
the current installer in `susi_installer` works on Debian stretch and above, and Ubuntu 18.04 and above. Please use it.

closed 2019-05-17 by norbusan