fossasia/susi_linux #453: The SUSI Server Doesn't auto boot

stealthanthrax, 2019-01-19:
#### Describe the bug
The SUSI Server doesn't autoboot and one has to manually boot it to work on the functionality,

#### To Reproduce
Steps to reproduce the behavior:
1. Start RasPi after installation
2. Ping localhost:8000

#### Expected behavior
To make it autoboot

#### Additional context
I'd like to take this up

hongquan, 2019-01-20:
It is automatically started, but because of some unknown reason, the startup is not always successful.

norbusan, 2019-05-17:
I think the problem is only with the boot not waiting for the server actually started. The recent changes I made in susi_installer and the systemd script should fix that. I suggest closing this.

norbusan, 2019-12-27:
That has been fixed for quite some time now, closing the issue

closed 2019-12-27 by norbusan