fossasia/susi_linux #457: To add a uninstall script for packages

stealthanthrax, 2019-01-29:
#### Is your feature request related to a problem? Please describe.
To add an uninstall script for the packages. it would help us ease the process of implementing the soft-reset feature.

#### Describe the solution you'd like
To create an uninstall script.

#### Describe alternatives you've considered
We can either delete all packages from the package from a soft-reset daemon itself or create a separate script. I would recommend deleting it from the same file.

#### Additional context
Add any other context or screenshots about the feature request here.

stealthanthrax, 2019-01-29:
@mariobehling @Orbiter @hongquan @cweitat , your suggestions?

stealthanthrax, 2019-02-03:
Okay @hongquan , I will try to look into packaging. Do you know of any good place to start reading?

hongquan, 2019-02-07:
Sorry, there is no good tutorial. I used to do it some years ago, but my knowledge may be outdated.

And before packaging, we need to redefine the app structure. Cannot let the app running inside _/home/pi_ like now. More things to concern, I already mentioned somewhere in GitHub issues.