fossasia/susi_linux #476: Enable Continuous Integration and deployment of susi_linux raspberry pi builds

alok760, 2019-04-07:
#### Is your feature request related to a problem? Please describe.
Currently after every change/update in the project the moderators have to manually build raspbian image with susi_linux and pi-gen. 

#### Describe the solution you'd like
automation of creating ready to use img files for susi_linux and implement continuous integration.

alok760, 2019-04-07:
i'm working on this.
progress can be checked here - https://github.com/alok760/raspbian_travis
and here - https://travis-ci.org/alok760/raspbian_travis

hongquan, 2019-04-08:
The description is not accurate. The image is already built automatically, not manually.

[image]

The only issue is that, it has to be run on real RPi hardware.

norbusan, 2019-05-17:
DOne via susi_installer and triggers from susi_linux, susi_python, closing this issue

closed 2019-05-17 by norbusan