fossasia/susi_linux #479: Add an Installation Script for Ubuntu

stealthanthrax, 2019-04-11:
#### Is your feature request related to a problem? Please describe.
Now, that we have a basic set of instructions(by @norbusan ) for setting up the smart speaker on an Ubuntu system. We should proceed by creating an installation script for the Ubuntu system as it will allow our development process to proceed much faster

#### Describe the solution you'd like
A bash script like install.sh

VishalSharma0309, 2019-04-11:
I would like to work on this issue.

norbusan, 2019-04-11:
I would prefer to have
- first a package of susi-server for Debian/Ubuntu - I am working on this and have already thought out the layout/location of files
- then we can look into the `susi_linux` (python/hotword etc) packaging/installation.

norbusan, 2019-04-12:
Those interested can try the `susi_server` installation via Debian package. The following description should work also on Ubuntu (but haven't tested!):

Add the following to `/etc/apt/sources.list.d/susi-ai-preining.list`:
```
deb http://www.preining.info/debian/susi.ai unstable main
```

After that you need to import by GPG key. EIther pick it from the Debian keyring, or simpler:
```
curl -fsSL https://www.preining.info/rsa.asc | apt-key add -
```
(as root, or calling the final apt-key with sudo).

Then do `apt-get update` and `apt-get install susi-server`. This will also install `susi-skill-data` package

A new user `_susiserver` will be created who runs the susi-server (all behind the scenes, nothing to worry).

Starting the server with `systemctl start susi-server` (as root or with sudo)

Please let me know your experience, problems, errors.

Things that are on my TODO:
- proper debian packaging: by now the gradle build pulls from the internet, this is a no-go in Debian. We should see whether all libraries are already packaged in Debian or not.
- stop scripts: not sure whether this is actually necessary, in my local experiments with `systemctl start/stop` I had no problems
- configuration: put `customized_config.properties` into `/etc/susi-server` and link it to the correct place

Well, but a start is done, looking forward to your feedback.

stealthanthrax, 2019-04-14:
@norbusan , why do we want to create a package for the server? Won't just a binary suffice?

norbusan, 2019-04-14:
@stealthanthrax very easy - if we want to *also* go to the desktop, we need to make it easy **and** a known process to install the necessary parts. Debian/Ubuntu users are used to installing a Debian package via *apt-get`, but not in manual installation.

Furthermore, proper setup is far out of reach for a normal user if you want a system wide susi server running (and you want, you don't want to install one susi server per user, right?): Do *you* know how to add a system user without password, and configure things that it works? Even I a Debian Developer had to read up before implementing the current packaging.

So yes, I think it is necessary to have packages, otherwise far less people will use it.

Last but not least, even for developers of the Python code without interest in the Java susi server can easily install the package and start hacking, without the need for fighting with the Java code.

stealthanthrax, 2019-04-15:
Alright @norbusan . Got it 👍

norbusan, 2019-04-26:
Installation on Ubuntu should now work using https://raw.githubusercontent.com/fossasia/susi_installer/development/install.sh
It allows installation on RPi, Ubuntu, and Debian.

I have only tested Debian and RPi, so experiences with Ubuntu would be appreciated.

norbusan, 2019-05-17:
Ubuntu install is working on 18.04 and 19.04, automatically tested on travis. Closing this issue

closed 2019-05-17 by norbusan