fossasia/susi_linux #482: Clean code to match Python coding style

hongquan, 2019-04-16:
### Describe the bug
Currently, many parts of our code don't follow Python coding style. Now, please clean it.
Please also group and arrange the `import` lines.

### Objective
- Follow PEP8 as much as possible. For line length, it can be 120 chars (PEP8 suggests 79, but I think that number is only relevant for old display device, while all of us now use wide screen monitor).
- Don't leave unused "import", unused variable. This can be detected with _pyflakes_ tool.

### Tools
This work doesn't need much brain. So we should leave it for tools. I recommend  _pycodestyle_, _pyflakes_. They help you find problematic lines, but not fix for you. Other than that, please install appropriate plugins to your code editor to integrate the tools.

This is how you can do in VS Code:
- Install [Python](https://marketplace.visualstudio.com/items?itemName=ms-python.python) extension.
- Add this to settings:

```json
{
    "python.linting.flake8Enabled": true,
    "files.trimTrailingWhitespace": true,
    "files.insertFinalNewline": true,
    "files.trimFinalNewlines": true
}
```

bhadreshpsavani, 2019-11-26:
May i work on this issue?

norbusan, 2019-11-27:
@bhadreshpsavani sure enough, go ahead and send PR

kiritka-jain, 2019-12-02:
it's been a while, @norbusan @stealthanthrax May I work on this ?

norbusan, 2019-12-02:
Everyone can work on that, just do it and send PRs...

yash04g, 2019-12-26:
I would like to work on this one. @stealthanthrax

PR #529 (2019-12-31): Feature issue 482
Fixes #482 

#### Checklist

-  [x] I have read the Contribution & Best practices Guide and my PR follows them.
-  [x] My branch is up-to-date with the Upstream master branch.
-  [ ] The unit tests pass locally with my changes
-  [ ] I have added tests that prove my fix is effective or that my feature works
-  [x] I have added necessary documentation (if appropriate)
-  [x] All the functions created/modified in this PR contain relevant docstrings.

#### Test Passing

- [ ] The SUSI Server must be building on the pi on bootup
- [ ] The hotword detection should have a decent ac [... cut]
closed 2019-12-31 by stealthanthrax