fossasia/susi_linux #495: All the images/gif in Readme are broken

stealthanthrax, 2019-05-12:
#### Describe the bug
After the folder architecture change of the project, links to all the demo images/gifs have been broken.

#### Expected behavior
To fix the links and updated the README

#### Additional context
Add any other context about the problem here.

norbusan, 2019-05-13:
@stealthanthrax Where are these demos? Where can I check them?

stealthanthrax, 2019-05-13:
@norbusan , I believe they were removed when the folder architecture was changed but I can add them from the history

norbusan, 2019-05-14:
I think the documents should be updated to point to `susi_installer`, where the images are better placed, IMHO.

stealthanthrax, 2019-05-15:
Okay @norbusan . That makes sense. I'll just remove the broken links from here.