fossasia/susi_linux #496: stopping sound hangs device

norbusan, 2019-05-17:
Currently, when I tell SuSI:
```
hey susi, play some music
```
it always plays the same music on the RPi. Then, when I say
```
hey susi, stop
```
it does stop the `vlc` process, but keeps it hanging somehow occupying the sound card. Asking Susi for something new does not play the detection bell sound, but hangs, too:
```
 1580 ?        TLl    0:09 /usr/bin/vlc -I dummy https://r4---sn-ogul7n7s.googlevideo.com/videoplayback?id=o-AGCHokAMEVnVZPxrSioJNZT
 1649 ?        S      0:00 play /home/pi/SUSI.AI/susi_linux/extras/detection-bell.wav
 1655 ?        S      0:00 sox -q -t mp3 - -d trim 0.1 reverse trim 0.07 reverse tempo 1.2 pitch 2 speed 1
```

This needs improvement.

hongquan, 2019-05-24:
Maybe we stopped VLC in a dirty way? Maybe we should tell VLC to stop the music first, then kill it?

norbusan, 2019-05-24:
I think this should be fixed by the susi_linux audi thread rework, later on.

alok760, 2019-05-26:
saying stop doesn't stop the song in my case
[image]

norbusan, 2019-06-07:
fixed with recent merge of sound rework

closed 2019-06-07 by norbusan