fossasia/susi_linux #501: More models for offline Speech Recognition

stealthanthrax, 2019-05-19:
#### Describe the bug
Currently, offline Speech Recognition only recognizes US English and more languages need to be supported.

#### Expected behavior
Configure more models from here https://sourceforge.net/projects/cmusphinx/files/Acoustic%20and%20Language%20Models/

#### Additional context
Reference for installation can be taken from here : https://github.com/Uberi/speech_recognition/blob/master/reference/pocketsphinx.rst#installing-other-languages

hongquan, 2019-05-19:
Good!

norbusan, 2019-05-19:
I think this is a very good idea, but we need setup interface for new languages, and isn't it some other service that by default is used for speech recognition?

stealthanthrax, 2019-05-22:
@norbusan , we use google stt when we are online and we use PocketSphinx when we are offline

himanshupathak21061998, 2019-09-20:
Hey @stealthanthrax I want to solve this issue. While using SUSI-AI offline I got following error
```
Error: missing PocketSphinx language data directory: "/usr/local/lib/python3.6/dist-packages/speech_recognition/pocketsphinx-data/en_US"
Internet Connection not available
```
and when I checked pocketsphinx-data it contains folder with name `en-us`.
I want to know is it some mistake done by me while setting up project.

stealthanthrax, 2019-09-20:
Hi @himanshupathak21061998 , I think this is a new bug which might have crept in while developing newer features.
I'll open an issue and you can work on fixing this first?

stealthanthrax, 2019-09-20:
Because , I don't think that there is a different way to install SUSI on your system instead.

norbusan, 2019-09-21:
@stealthanthrax @himanshupathak21061998 interesting. The `en_US` comes from proper language support, so we use locale names. Back then I changed the invocation of the sphinx recognizer to
```
recognizer.recognize_sphinx(audio, language=susi_config["language"])
```
where `susi_config["language"]` contains a locale style string (ll_LL). We need to convert this to one of the supported languages of pocket sphinx on the fly.

Best would be further to have the code check which languages are installed, and fall back to english if the requested language is not available.

It looks like an easy few line Python hack in `susi_linux/main/states/recognizing_state.py`, anyone wanting to take that? Otherwise I do it later on.