fossasia/susi_linux #502: Add support for other debian based distros like mint

alok760, 2019-05-24:
#### Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
- Currently, the installer only supports Debian, Ubuntu 18.04 and 19.04

#### Describe the solution you'd like
A clear and concise description of what you want to happen.
- Add support for other Debian based distributions such as Linux Mint, Zorin OS, Elementary OS

This is an epic issue because it requires multiple pull request for each OS, Please suggest other OSs also

norbusan, 2019-05-24:
Mint should already be working (because @Orbiter pushed me to that on the gitsatellite conference), but the others I have no idea about.What I would need from those is the output of `lsb_release -r` and `lsb_release -i`, and what kind of packages from 
```
DEBDEPS="
  git openssl wget python3-pip sox libsox-fmt-all flac
  libportaudio2 libatlas3-base libpulse0 libasound2 vlc-bin vlc-plugin-base
  vlc-plugin-video-splitter python3-cairo python3-flask flite
  default-jdk-headless pixz udisks2 python3-requests python3-service-identity
  python3-pyaudio python3-levenshtein python3-pafy python3-colorlog
  python3-watson-developer-cloud ca-certificates
"
```
and `python3-alsaaudio` are available.

norbusan, 2019-05-25:
This should actually be an issue of `susi_installer`. Unfortunately the "Transfer issue" does not work for me.

I have added support for Mint in a branch.

norbusan, 2019-06-30:
Debian, Ubuntu, Mint is now supported. If there are others, please open a new issue

closed 2019-06-30 by norbusan