fossasia/susi_linux #508: on saying "stop" smart speaker says "I don't have an answer to this"

alok760, 2019-05-30:
#### Describe the bug
A clear and concise description of what the bug is.
- whenever music is playing, on saying "stop" the smart speaker says "I don't have an answer to this" and then stops the song

#### To Reproduce
Steps to reproduce the behaviour:
Smart Speaker - 
1. Susi, Play a song
2. Susi, Stop

#### Expected behavior
A clear and concise description of what you expected to happen.
- either stop the song silently or say "stopped!"

alok760, 2019-05-30:
I am working on this.

norbusan, 2019-05-31:
It is trivial, in `busy_state.py` there is a long list of
```
if 'xxx' in reply.keys():
   ...
```
These are no `elif` but just `if ...`
There is a test
```
            if 'answer' in reply.keys():
                logger.info('Susi: %s', reply['answer'])
                lights.off()
                lights.speak()
                self.__speak(reply['answer'])
                lights.off()
            else:
                lights.off()
                lights.speak()
                self.__speak("I don't have an answer to this")
                lights.off()
```
but when recognizing stop the only return value is
```
{'stop': <susi_python.models.StopAction object at 0x7f8ac72b98d0>}
```
so the `I don't have any answer to this` is spoken.

Later on there is
```
            if 'stop' in reply.keys():
                player.stop()
```
(in my branch).

That whole part of code needs a serious rewrite ...

norbusan, 2019-05-31:
Furthermore, the `pause` action is broken and returns:
```
[<susi_python.models.MediaAction object at 0x7f7afe4ef2b0>]
2019-05-31 15:06:53 ERROR:main.states.busy_state: Got error: 'MediaAction' object has no attribute 'type'
Error: None 
```

norbusan, 2019-05-31:
Ok, fixed `susi_python`, it was broken for MediaAction and didn't set the `type`.

norbusan, 2019-05-31:
Fixed pause, resume, etc etc in my sound branch, see commit
https://github.com/norbusan/susi_linux/commit/6048ace543bb89595e647fa45e2fce36010340ff

closed 2019-06-04 by alok760