fossasia/susi_linux #510: susi_linux raises "cannot find mixer exception" on the smart speaker

alok760, 2019-06-04:
#### Describe the bug
A clear and concise description of what the bug is.
On starting susi_linux :
```
local server is down
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.5/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.5/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/pi/SUSI.AI/susi_linux/main/__init__.py", line 3, in <module>
    from .states import SusiStateMachine
  File "/home/pi/SUSI.AI/susi_linux/main/states/__init__.py", line 3, in <module>
    from .susi_state_machine import SusiStateMachine
  File "/home/pi/SUSI.AI/susi_linux/main/states/susi_state_machine.py", line 14, in <module>
    from .busy_state import BusyState
  File "/home/pi/SUSI.AI/susi_linux/main/states/busy_state.py", line 11, in <module>
    from ..speech import TTS
  File "/home/pi/SUSI.AI/susi_linux/main/speech/TTS.py", line 14, in <module>
    from ..player import player
  File "/home/pi/SUSI.AI/susi_linux/main/player/__init__.py", line 5, in <module>
    from vlcplayer import vlcplayer
  File "/home/pi/SUSI.AI/susi_installer/vlcplayer/__init__.py", line 6, in <module>
    from hwmixer import mixer
  File "/home/pi/SUSI.AI/susi_installer/hwmixer/__init__.py", line 41, in <module>
    mixer = HwMixer()
  File "/home/pi/SUSI.AI/susi_installer/hwmixer/__init__.py", line 15, in __init__
    raise Exception('Cannot find mixer')
Exception: Cannot find mixer
```

#### To Reproduce
Steps to reproduce the behavior:
1. run susi_linux on smart speaker

norbusan, 2019-06-04:
Yeah, that is known. You have to link `hwmixer` directory from `susi_installer` to `susi_linux`, similar to `susi_python`. I will rework this (module location and loading).

alok760, 2019-06-04:
@norbusan I did link the hwmixer directory 
it was happening due to-
alsaaudio.mixers() on pi only shows "PCM"
I  fixed it by adding these lines to `/etc/asound.conf`, not sure if this is the ideal solution
```
pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
  }
}

norbusan, 2019-06-04:
Ahh, interesting.
No, don't do edit /etc/asound.conf. We need to fix it in hwmixer. Can you send the output of alsasound.mixers() here please

alok760, 2019-06-04:
> output of alsasound.mixers()
```
>>> alsaaudio.mixers()
['PCM']
```

norbusan, 2019-06-04:
Hmm, that is strange ... that should no happen. The respeaker hat should (at least when I tested it) give 'Speaker' as answer. Did you set up the wlan of the speaker?

norbusan, 2019-06-04:
After proper setup of the respeaker the output should be:
```
pi@raspberrypi:~/SUSI.AI/seeed-voicecard $ python3 
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import alsaaudio
>>> alsaaudio.mixers()
['Headphone', 'Headphone Playback ZC', 'Speaker', 'Speaker AC', 'Speaker DC', 'Speaker Playback ZC', 'PCM Playback -6dB', 'Mono Output Mixer Left', 'Mono Output Mixer Right', 'Playback', 'Capture', '3D', '3D Filter Lower Cut-Off', '3D Filter Upper Cut-Off', 'ADC Data Output Select', 'ADC High Pass Filter', 'ADC PCM', 'ADC Polarity', 'ALC Attack', 'ALC Decay', 'ALC Function', 'ALC Hold Time', 'ALC Max Gain', 'ALC Min Gain', 'ALC Mode', 'ALC Target', 'DAC Deemphasis', 'DAC Mono Mix', 'DAC Polarity', 'Left Boost Mixer LINPUT1', 'Left Boost Mixer LINPUT2', 'Left Boost Mixer LINPUT3', 'Left Input Boost Mixer LINPUT1', 'Left Input Boost Mixer LINPUT2', 'Left Input Boost Mixer LINPUT3', 'Left Input Mixer Boost', 'Left Output Mixer Boost Bypass', 'Left Output Mixer LINPUT3', 'Left Output Mixer PCM', 'Noise Gate', 'Noise Gate Threshold', 'Right Boost Mixer RINPUT1', 'Right Boost Mixer RINPUT2', 'Right Boost Mixer RINPUT3', 'Right Input Boost Mixer RINPUT1', 'Right Input Boost Mixer RINPUT2', 'Right Input Boost Mixer RINPUT3', 'Right Input Mixer Boost', 'Right Output Mixer Boost Bypass', 'Right Output Mixer PCM', 'Right Output Mixer RINPUT3']
>>> 
```

alok760, 2019-06-05:
@norbusan I'll try with a fresh install and update

norbusan, 2019-06-05:
Hi @alok760 
I see the same now, it seems that *rebooting* the system brings it into a strange state, and the respeaker is not properly initialized.

I have no idea what is going on wit the `dtparam` and `dtoverlay` calls, but this seems to be new.

norbusan, 2019-06-05:
Something is strange, I submitted an issue to respeaker: https://github.com/respeaker/seeed-voicecard/issues/166

There was a kernel/firmware update of raspbian in mid-May, this might have triggered these problems we see now.

norbusan, 2019-06-07:
I don't know what has changed, but with the current setup the mixer is called PCM, I adjusted the code. CLosing again

closed 2019-06-07 by norbusan