fossasia/susi_linux #513: Susi does not leave recognizing state

norbusan, 2019-06-07:
Susi remains indefinitely in recognizing state, even if nothing is said. We should add a timeout of say 10sec after which it transitions back to idle state.

stealthanthrax, 2019-11-06:
@norbusan , has this issue been solved?
I'd like to take this is up, if pending.

norbusan, 2019-11-06:
Yes please, that would be great! It is still the case, we need a timeout

stealthanthrax, 2019-11-06:
@norbusan , what do you suggest? SUSI to transition directly back to the IDLE state or through the Error State and then reaching the IDLE state?

norbusan, 2019-11-07:
Directly into idle state should be fine. Alexa gives a short "to-to" (very short, falling tone) when it stops recognizing. If we can have something like this it would be nice, too.

PR #527 (2019-11-15): Added a 10 second timeout in the recognizing state
Fixes #513 
This should work in principle, but since I haven't tested it on my speaker hence still WIP.

#### Checklist

- [x] I have read the Contribution & Best practices Guide and my PR follows them.
- [x] My branch is up-to-date with the upstream master branch.
- [x] The unit tests pass locally with my changes
- [x] I have added tests that prove my fix is effective or that my feature works
- [x] I have added necessary documentation (if appropriate)
- [x] All the functions created/modified in this PR contain relevant docstrings.

#### Test  [... cut]
closed 2019-11-15 by norbusan