fossasia/susi_linux #514: Create a server for audio remote/controller in co-ordination with the Android App

stealthanthrax, 2019-06-08:
#### Is your feature request related to a problem? Please describe.
To add a way to control the music playback from mobile devices

#### Describe the solution you'd like
To create a flask server that would act as a host/intermediate server between the Android client and the music server running on the raspberry pi.

#### Describe alternatives you've considered
We could've used only the sound server running on the device but exposing the endpoints to every device on the network is not a wise decision.

So basically, we will create a proxy server on the pi which will act as an intermediate between the mobile and the smart speaker

#### Additional context
Add any other context or screenshots about the feature request here.

stealthanthrax, 2019-06-08:
SUSI Andorid Link: https://github.com/fossasia/susi_android/issues/2251

norbusan, 2019-06-09:
@stealthanthrax Why would you want a separate server which just proxies? If you add authentication to a flask server, you can add it to the current soundserver directly. Security wise there is no difference?

stealthanthrax, 2019-06-10:
@norbusan , the flask server will still be accessible to every device on the network and the request headers can be analyzed by simple mitm techniques. But since the two servers can be will be communicating internally,  we will be hiding the endpoints of the sound server and hence will stop other devices on the same network from fiddling with the settings.

norbusan, 2019-06-10:
@stealthanthrax So you mean that only parts of the end points of the sound server are forwarded via the second plask server?

What kind of end points of the current sound server do you **NOT** want to forward?

hongquan, 2019-06-11:
Sorry. I lost the track. Which is sound server and which is Flask server? I remember we have something running at port 4040. What is it?

stealthanthrax, 2019-06-11:
@hongquan , this is the server https://github.com/fossasia/susi_installer/blob/development/soundserver/soundserver.py

stealthanthrax, 2019-06-11:
@norbusan , now that I think about it, I don't think that we should create a proxy server , since already the codebase is open source and all the endpoints are already public, adding authentication to the existing server is a better idea.

hongquan, 2019-06-11:
I agree with adding authentication to sound server.

Also, the sound server is using "GET" method to trigger actions. I feel that this is very weird.
Normally, "GET" is to read data, and for anything that causes modification of data, "POST", "PUT" should be used.

norbusan, 2019-06-11:
@stealthanthrax well, if you really want to expose only *some* end points, then you can still do that. You can tell the soundserver flask to only listen to localhost, but not any other connection. Thus, it could only be used directly by susi_linux and the gateway server. The other server with authentication would only forward a restricted set of end points. 

So it is possible, even if the end points are known. That is the reason why I asked which end points you want to hide.

@hongquan well, using get allows easy conrol via the browser, I can test the server by typing in some command into the browser windows, or send wget/curl requests.

hongquan, 2019-06-12:
@norbusan Defining as "POST/PUT" will make you difficult to test with browser, but it is still easy to test with CLI tools.

One of the reason that people don't use `GET` for action which modify data, because it is easy for attack, by tricking browser to call it.

For example, you have _http://example.com/api/delete-user_, which is called by `GET`. The attacker just put this code on some website

```html
<img src='http://example.com/api/delete-user'>
```
and will trick browser to do "delete user" on your website.

norbusan, 2019-06-12:
Ok, fine with me!

norbusan, 2019-06-12:
Ok, let us support POST and PUT, but not GET. https://github.com/fossasia/susi_installer/pull/32

norbusan, 2019-06-30:
Closing this now that the server is in the image and activated

closed 2019-06-30 by norbusan