fossasia/susi_linux #521: Timed Actions

Orbiter, 2019-06-24:
As a user I want to get timed actions: these are actions which are activated by a given time.
Using timed actions it is possible to make not only "a timer" possible, but it is also possible to activate any kind of actions for a given time.

A broad range of use cases are possible:
- a simple timer which activates a voice message and a sound (two actions, both timed)
- a count-down (many actions, all messages with time-replated text like "20 seconds left")
- a frequent/regular reminder (at 12:00 SUSI says: "it's time to take your pills")
- an alarm clock to wake you up (one music play action at a specific time)
- project planning (family to leave the house at 11:00 - a set of message from "brush your teeth" to "put on your shoes" etc.)

#### Describe the solution you'd like
We need a scheduler which hosts SUSI actions which have a "timed" object attached. Such an action is not performed at the time when it arrives from the susi_server, instead it is performed if it is due in the scheduler. The due-date is arrived if the local date is greater than the date as given in the "timed" object.

To align the time of the server and the time of the client, the `query_date` as given in the susi response is used. This time is represented in the ISO8601 format and is time-zone independent. 

The "timed" object has the following form:
```
"timed" = {
  "due" = <ISO8601 date>,
  "delay" = <time in milliseconds>
}
```
Here the `delay` value is only used for debugging purpose, it is useful if the timed action was used for a timer; then the delay should represent the timer value in milliseconds.

A full susi response with timed actions would look like:

```

  "query": "wake me up at 6:30 with metallica",
  "query_date": "2019-06-24T13:36:25.490Z",
  "answer_date": "2019-06-24T13:36:25.752Z",
  "answers": [
    {
      "actions": [
        {
          "type": "answer",
          "expression": "Good Morning! Playing Metallica..",
          "timed" = {
            "due" = "2019-06-25T05:00:00.000Z"
          }
        },
        {
          "identifier": "/a-jameshetfield/metallica-nothing-else-matters",
          "identifier_type": "soundcloud",
          "type": "audio_play",
          "timed" = {
            "due" = "2019-06-25T05:00:05.000Z"
          }
        }
      ]
    }
  ]
}
```

#### Important
It is NOT sufficient to use the local time of the client and just add up the delay time in milliseconds to compute a local due-time. We want that client can activate actions based on exact times as defined by the skill. This should be as exact as possible to make it feasible that this can be used for machine-to-machine interaction once several machines are using SUSI and communicate using timed events.
Therefore 
- the given due-time must be used as activation moment for the action
- the client must perform a validation check of the local clock with the given `answer_date`: if the local date derives too much from the given `answer_date` the local clock must be corrected to make timed actions possible and exact. Think of this like lives depend on it!!

stealthanthrax, 2019-07-23:
@Orbiter @norbusan  , we can create a systemd.timer and .service on the fly and activate the service for every action that the user calls and disable the service on completion?

norbusan, 2019-07-23:
@stealthanthrax For that we would need a way to inject actions into susi_linux main loop.but I think it would be better to have a scheduler in the python code that wakes up every sec and checks for events to be executed.

hongquan, 2019-07-24:
Agree. Because of the vast variation of need for action/schedule, it should be implemented in our Python code. We need to create a candidate process to perform the action/job. Then from our main project, we send the task to that process, include the time we want it to be done.

We can use https://github.com/rq/rq-scheduler#periodic--repeated-jobs for this.

norbusan, 2019-07-24:
@hongquan yes, I agree completely that we should do this in our own code. 

Concerning your suggestion of rq-scheduler, this requires redis package. Isn't that a bit too heavy? I haven't really tried it out, though. Do you have experience with that?

hongquan, 2019-07-25:
@norbusan No worries. Redis is one of the lightest software: https://packages.ubuntu.com/disco/redis-server

stealthanthrax, 2019-07-26:
This makes much more sense.
> have a scheduler in the python code that wakes up every sec and checks for events to be executed

stealthanthrax, 2019-07-26:
@hongquan , we can also use something like rabbitmq and celery for task scheduling

hongquan, 2019-07-27:
@stealthanthrax Those names are more wel-known, but they are too fat and slow, especially on an ARM computer like Raspberry Pi.
- RabbitMQ is written in Erlang, it means that you have to install an Erlang Virtual Machine.
- Celery tries to support many backend (RabbitMQ, Redis, MongoDB etc...), many serialization formats (pickle, json, msgpack etc.), so its code is too thick.

norbusan, 2019-07-27:
What about https://github.com/dbader/schedule/ which is also on pypi, very lightweight...

hongquan, 2019-07-27:
`schedule` looks OK.

alok760, 2019-07-31:
> What about https://github.com/dbader/schedule/ which is also on pypi, very lightweight...

looks promising.
I'll start working on this

hongquan, 2019-08-01:
Please note that, because [schedule](https://github.com/dbader/schedule/) works alone (without a message queue like Redis), it can only do one task at once and will miss any command you send to it, when it is busy.

In the case of other solutions (RQ, Celery), your command is put to a queue in Redis/RabbitMQ, and the worker can still get the command later, after it finishes its job.

norbusan, 2019-08-05:
@hongquan @alok760 that is a good point, though it shouldn't be busy too long, and only kick of an action in the main loop, but not waiting for completion. That way this window is really small, and other smart speakers will have similar issues.

Still, it is good to keep in mind, thanks!!

PR #524 (2019-08-24): build plan action support
Fixes #521

#### Checklist

- [x] I have read the Contribution & Best practices Guide and my PR follows them.
- [x] My branch is up-to-date with the Upstream master branch.
- [ ] The unit tests pass locally with my changes
- [ ] I have added tests that prove my fix is effective or that my feature works
- [ ] I have added necessary documentation (if appropriate)
- [ ] All the functions created/modified in this PR contain relevant docstrings.

#### Test Passing

- [ ] The SUSI Server must be building on the pi on bootup
- [ ] The hotword detection should have a decent a [... cut]
closed 2019-08-24 by norbusan