fossasia/susi_linux #528: Adding sensible error messages when susi crashes

stealthanthrax, 2019-11-22:
#### Is your feature request related to a problem? Please describe.
Currently, SUSI only sends out one error message when it crashes.

#### Describe the solution you'd like
Different error messages depending on the issue.

#### Additional context
Only changes in the state machine required.

norbusan, 2019-11-23:
When SUSI crashes, you mean the susi_linux process? But when it crashes, how could it give sensible messages? Ok, we could create a state file, clean it up on proper shutdown, and when systemd restarts the susi_linux process and it finds the state file, it could tell the user that susi linux has crashed, but how do you intend to get details?

stealthanthrax, 2019-11-23:
@norbusan , I meant for example when there is an issue with the internet connection or it is a problem with deploying Java server on the Pi . These errors are handled in the `error_state` file. But only one kind of sound played when it enters the error state.
Instead of that, we should play different sounds for easier debugging.

norbusan, 2019-11-25:
Ok, so you mean for TTS or STT, when the connection timed out.
What about problems communicating with the onboard susi_server, can we detect that, too?

stealthanthrax, 2019-11-25:
@norbusan , yes we handle those too. I think we log the errors somewhere as of now.