fossasia/susi_linux #530: Continuous repetition of "not able to recognize" error messages

pranavkarthik10, 2019-12-17:
After installation of susi_installer on Ubuntu, I configured and ran susi_linux using python3. Right when it started even though I had made no noise, it began a continuous repetition of 'Sorry, I couldn't recognize' messages were said immediately before stopping after a few seconds.

I also found that the sensitivity of the speech-to-text system could be minimized because of it catching up on the slightest sounds nearby. A possible starting phrase such as 'Hey Susi' or just 'SUSI'  before stating any message from the user's end could be used to not activate the assistant unnecessarily.

norbusan, 2019-12-18:
We unfortunately don't have control over the microphone settings on the desktop for now. You need to adjust the correct microphone and the recording level manually.

Can you please stop susi-python process, and start it by hand with
```
cd ~/SUSI.AI/susi_python/
python3 -m main -v -v
```
and attach the output here? Thanks

pranavkarthik10, 2019-12-19:
Attached output here: https://pastebin.com/cSe2ZyyT

norbusan, 2019-12-19:
A few things:
- you don't have the susi server running, right? susi_linux says `local server is down` in the first line
- what is contained in `~/SUSI.AI/config.json`

Then please re-test with the following commands:
```
cd ~/SUSI.AI
PYTHONPATH=/home/norbert/SUSI.AI/susi_linux:$PYTHONPATH
export PYTHONPATH
python3 -m main -v -v
```

But yeah, I see some problems here myself, too ...
I see that

pranavkarthik10, 2019-12-23:
Unfortunately my computer in which I was running the VM for linux is currently not working well. Sorry for any delays.

norbusan, 2019-12-27:
I have fixed these problems in the `susi_linux` code, thus closing the issue

closed 2019-12-27 by norbusan