fossasia/susi_linux #533: Update Readme and evaluate repo name

mariobehling, 2020-01-16:
The Readme is not yet formulated very clearly. At the beginning of the Readme there should be information what this repository contains.

The Readme should also contain information on what specifically is the goal of this repo compared to other repositories of the SUSI.AI project.

The name of the repo might indicate to an outsider that the repository contains all files and Linux components to build a SUSI.AI Linux distro, but the actual SUSI.AI image for smart speakers is hosted in the installer repository here: https://github.com/fossasia/susi_installer/releases
Therefore what is exactly the core idea of this repository? What would be the best name of this repo, should it be changed?

norbusan, 2020-01-18:
Concerning the names, I thought about `susi_voice_component` - that is probably the most fitting name ;-)

snitin315, 2020-01-20:
I will work on this issue.